ni/nimi-python #2075: Add Python 3.13 support and loosen protobuf version constraint

bkeryan, 2025-01-28:
### Description of issue

The current stable release of Python is version 3.13, which was released on 10-7-2024.

There are versioning conflicts between nimi-python's `grpc` extra and Python 3.13:
- nimi-python's `grpc` extra requires `protobuf>=4.21.6,<5.0`.
- The first version of `grpcio` and `grpcio-tools` with binary wheels for Python 3.13 is [version 1.67.0](https://pypi.org/project/grpcio/1.67.0/)
- `grpcio-tools` version 1.67.0 requires `protobuf` version `<6.0dev,>=5.26.1`.

This causes issues when upgrading `ni-measurement-plugin-sdk-service`'s dev/test dependencies, since it depends on `grpcio-tools` for gRPC codegen and several nimi-python packages for testing.

```
PS C:\dev\measurement-plugin-python\packages\service> poetry lock --no-update
Resolving dependencies... (4.9s)

Because no versions of niswitch match >1.4.4,<1.4.5 || >1.4.5,<1.4.6 || >1.4.6,<1.4.7 || >1.4.7,<1.4.8 || >1.4.8
 and niswitch[grpc] (1.4.4) depends on protobuf (>=4.21,<5.0), niswitch[grpc] (>=1.4.4,<1.4.5 || >1.4.5,<1.4.6 || >1.4.6,<1.4.7 || >1.4.7,<1.4.8 || >1.4.8) requires protobuf (>=4.21,<5.0).
And because niswitch[grpc] (1.4.5) depends on protobuf (>=4.21,<5.0), niswitch[grpc] (>=1.4.4,<1.4.6 || >1.4.6,<1.4.7 || >1.4.7,<1.4.8 || >1.4.8) requires protobuf (>=4.21,<5.0).
And because niswitch[grpc] (1.4.6) depends on protobuf (>=4.21,<5.0)
 and niswitch[grpc] (1.4.7) depends on protobuf (>=4.21,<5.0), niswitch[grpc] (>=1.4.4,<1.4.8 || >1.4.8) requires protobuf (>=4.21,<5.0).
And because niswitch[grpc] (1.4.8) depends on protobuf (>=4.21.6,<5.0)
 and grpcio-tools (1.67.0) depends on protobuf (>=5.26.1,<6.0dev), niswitch[grpc] (>=1.4.4) is incompatible with grpcio-tools (1.67.0).
So, because ni-measurement-plugin-sdk-service depends on both niswitch[grpc] (>=1.4.4) and grpcio-tools (1.67.0), version solving failed.
```

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

N/A, can't install

### Steps to reproduce issue

1. Download https://github.com/ni/measurement-plugin-python/tree/users/bkeryan/python-3.13
2. cd to packages/service
3. `poetry lock --no-update`

ni-jfitzger, 2025-02-05:
@bkeryan For some reason, the pb2 files generated with grpcio-tools 1.67.0 resulted in a protobuf dependency of 5.27.2, despite [the clear declaration of compatibility with 5.26.1](https://github.com/grpc/grpc/blob/e06ad82c3fdde9ec6598d314998d88e848f4c577/tools/distrib/python/grpcio_tools/setup.py#L346).

```
from google.protobuf.internal import builder as _builder
_runtime_version.ValidateProtobufRuntimeVersion(
    _runtime_version.Domain.PUBLIC,
    5,
    27,
    2,
    '',
    'nidcpower.proto'
)
```

I had to set my tests env protobuf dependency to 5.27.2 to get the unit tests to pass. With 5.26.1, I ran into import errors.

bkeryan, 2025-02-05:
I think you should generate code with the oldest supported Python and grpcio-tools version, check the generated code into version control, and then test against newer versions.

https://github.com/ni/nimi-python/pull/2080#discussion_r1943453414

closed 2025-02-18 by ni-jfitzger