ni/nimi-python #2096: Make it easier to contribute to this project by replacing WSL and GNU Make with Python

bkeryan, 2025-05-01:
The contributing workflow for this project on Windows requires installing WSL and the Linux packages for GNU Make, Info-ZIP, and Python. For developers using `pyenv`, additional configuration is required to hide the Windows `pyenv` and Python versions from the Linux guest. These are rather heavyweight dependencies for contributing to a pure-Python package that does not build any C extensions.

The current nimi-python developer workflow involves running Tox, which calls GNU Make as needed. If you replaced the makefiles with Python scripts, this developer workflow would not need to change. Running Tox would call the Python scripts instead of GNU Make.

Other NI open-source Python projects such as https://github.com/ni/nidaqmx-python , https://github.com/ni/measurement-plugin-python , and https://github.com/ni/nisync-python perform codegen using Python scripts, which has worked out well. Because the projects are hosted in Git, it doesn't really matter whether the codegen updates the file timestamps every time you run it, and the codegen is fast enough that it wouldn't really benefit from skipping unchanged files.

If `\r\n` vs. `\n` line endings are an issue, configure `.gitattributes` to automatically convert the appropriate files.

(BTW: I have experienced WSL-specific networking issues on my dev machine, but I also have a bare-metal Linux desktop PC, so I don't **have** to use WSL.)

marcoskirsch, 2025-05-02:
> and the codegen is fast enough that it wouldn't really benefit from skipping unchanged files.

Debatable.

But I agree 100% that even if the incremental build goes away, removing the non-trivial barrier of entry of installing and having to understand GNU Make makes this worthy.

There are, should we be inclined to look into it, pure Python alternatives such as [DoIt](https://pydoit.org). But I kind of lean towards KISS and not adding a dependency on something this crucial.

ni-jfitzger, 2025-05-02:
>For developers using pyenv, additional configuration is required to hide the Windows pyenv and Python versions from the Linux guest.

I think we have some steps for this documented somewhere in an internal work item. It may be easier to just list those in the CONTRIBUTING.md.

>and the codegen is fast enough that it wouldn't really benefit from skipping unchanged files.

That's easy to say when you haven't had to sit through codegen (dozens of?) dozens of times, like I have. Maybe it would be faster in Python; I don't know.

I'm not sure that eliminating GNU Make frees us from the need for WSL. We still need tox. How easy is that to install on Windows?

---

This may be worth doing. Quite a large undertaking, though.

bkeryan, 2025-05-02:
> > and the codegen is fast enough that it wouldn't really benefit from skipping unchanged files.
>
> That's easy to say when you haven't had to sit through codegen (dozens of?) dozens of times, like I have. Maybe it would be faster in Python; I don't know.

Sorry, I was referring to the codegen in the other projects, not this one.
- nidaqmx-python codegen takes 3 seconds: https://github.com/ni/nidaqmx-python/actions/runs/14714570033/job/41294863527
- measurment-plugin-python codegen takes 1 second: https://github.com/ni/measurement-plugin-python/actions/runs/14786235898/job/41515039632

If your makefiles invoke `python` many times, changing to a single `python` invocation may speed it up significantly by reducing the overhead of importing modules and loading C extensions.

> I'm not sure that eliminating GNU Make frees us from the need for WSL. We still need tox. How easy is that to install on Windows?

Tox is a pure Python package so you install it like other Python packages. 

In projects that use Poetry + Tox, you run `poetry install` and then `poetry run tox`. https://github.com/ni/nidaqmx-python/blob/master/tox.ini shows how to configure Tox to use poetry to install locked dependencies.

bkeryan, 2025-05-05:
BTW, another way to deploy GNU make for development is to use [Visual Studio Code Dev Containers](https://code.visualstudio.com/docs/devcontainers/create-dev-container). This would still require WSL, plus Docker/Rancher/whatever, but it would allow you to specify which Linux packages to install and they would be isolated from the user's default WSL environment.